signal11/hidapi #336: Access denied when sending / reading data to / from Steam Controller

iUltimateLP, 2017-06-07:
Hey, thanks for hidapi!
I want to communicate with the Steam controller, but I have a problem here.    
The Steam Controller is identified by Windows in the device managment console as a "HID-conform Mouse". The driver it uses on Windows is "msmouse.inf", the service "mouhid".   
Now, I can open the device using hidapi when I'm using this (https://gist.github.com/cactrot/7242364) patch, but I get an access denied error when I want to send or read data from/to the controller.  
Example with reading:
```cpp
	res = hid_read(controller, buf, 17);
	printf("Read: %d. Error: %ls\n", res, hid_error(controller));
```
Where res is -1 and the error is "Access denied". When I try to send data:
```cpp
	unsigned char dataBlob[64] = { 0x8f,
		0x07,
		0x00, //Trackpad select : 0x01 = left, 0x00 = right
		0xff, //LSB Pulse High Duration
		0xff, //MSB Pulse High Duration
		0xff, //LSB Pulse Low Duration
		0xff, //MSB Pulse Low Duration
		0xff, //LSB Pulse repeat count
		0x04, //MSB Pulse repeat count
		0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
		0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
		0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
		0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 };

	double frequency = midiFrequency[note];
	double period = 1.0 / frequency;
	uint16_t periodCommand = period * 495483.0;

	//Compute number of repeat. If duration < 0, set to maximum
	uint16_t repeatCount = (duration >= 0.0) ? (duration / period) : 0x7FFF;

	dataBlob[2] = bRightHaptic ? 0x00 : 0x01;
	dataBlob[3] = periodCommand % 0xff;
	dataBlob[4] = periodCommand / 0xff;
	dataBlob[5] = periodCommand % 0xff;
	dataBlob[6] = periodCommand / 0xff;
	dataBlob[7] = repeatCount % 0xff;
	dataBlob[8] = repeatCount / 0xff;

	int r;
	r = hid_write(controller, dataBlob, sizeof(dataBlob));
	if (r < 0)
	{
		printf("Unable to send note. Error: %ls\n", hid_error(controller));
	}
```
I also get access denied. What is the problem here?  
Thanks for your help :)

signal11, 2017-06-07:
You can't read or write reports from or to keyboards and mice on Windows. The OS won't allow it.

iUltimateLP, 2017-06-07:
Oh, how'd I go in communicating with it then?

signal11, 2017-06-07:
Good question. Look for some windows-specific APIs. I remember something about low-level mouse events.

iUltimateLP, 2017-06-07:
I found a way using libusb, thanks anyway!

closed 2017-06-07 by iUltimateLP

signal11, 2017-06-07:
Hmm, that actually surprises me as libusb on Windows uses the platform HID libraries as its back-end for HID devices on Windows, which I would expect to have the same limitation. It's been a while since I've looked at what they have though. I'm glad you got it working.

kosme, 2019-12-19:
@iUltimateLP I am facing the same problem, could you share how you fixed this issue?