signal11/hidapi #337: MacOSX 10.12: Library crashes when unplug device just at the right time

chalio168ti, 2017-06-08:
Mac OS X 10.12.5

Here the stack trace:

> Thread 57 Crashed:
0   libsystem_kernel.dylib        	0x00007fffb4c42d42 __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fffb4d30457 pthread_kill + 90
2   libsystem_c.dylib             	0x00007fffb4ba8420 abort + 129
3   libjvm.dylib                  	0x0000000102823f5f os::abort(bool) + 25
4   libjvm.dylib                  	0x000000010294ae9e VMError::report_and_die() + 2304
5   libjvm.dylib                  	0x0000000102825b8a JVM_handle_bsd_signal + 1131
6   libjvm.dylib                  	0x0000000102821def signalHandler(int, __siginfo*, void*) + 47
7   libsystem_platform.dylib      	0x00007fffb4d23b3a _sigtramp + 26
8   ???                           	000000000000000000 0 + 0
9   com.apple.framework.IOKit     	0x00007fffa0f0a0f3 __IOHIDDeviceNotification + 187
10  com.apple.framework.IOKit     	0x00007fffa0ed86b1 IODispatchCalloutFromCFMessage + 308
11  com.apple.CoreFoundation      	0x00007fff9ef653c3 __CFMachPortPerform + 291
12  com.apple.CoreFoundation      	0x00007fff9ef65289 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE1_PERFORM_FUNCTION__ + 41
13  com.apple.CoreFoundation      	0x00007fff9ef65201 __CFRunLoopDoSource1 + 465
14  com.apple.CoreFoundation      	0x00007fff9ef5d105 __CFRunLoopRun + 2405
15  com.apple.CoreFoundation      	0x00007fff9ef5c544 CFRunLoopRunSpecific + 420
16  libhidapi.dylib               	0x000000010ece4232 read_thread + 274
17  libsystem_pthread.dylib       	0x00007fffb4d2d93b _pthread_body + 180
18  libsystem_pthread.dylib       	0x00007fffb4d2d887 _pthread_start + 286
19  libsystem_pthread.dylib       	0x00007fffb4d2d08d thread_start + 13

Seems to me that the IOKit it's not doing a great job on handling the Source's RunLoop execution (due to a possible incoming report) when the device has been removed from the System.

I'm not sure how to prevent this collision if the Source it's been executed (just at the right time) when the device it's been removed.

Any ideas of how to synchronize/workaround this are more than welcome, thanks.

NOTE: My App as you can tell from the stack trace, it's running inside of a JVM.

tresf, 2017-06-08:
> it's running inside of a JVM.

@chalio168ti If you're doing this from Java, you may want to look at this thread... https://github.com/qzind/tray/issues/137#issuecomment-277701230

chalio168ti, 2017-06-09:
@tresf thanks for the info. Unfortunately that specific fix addresses a crash during hid_enumerate() (stack from the issue https://github.com/qzind/tray/issues/137#issue-189579663):

> Stack: [0x000070000de54000,0x000070000df54000],  sp=0x000070000df51580,  free space=1013k
Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
C  [CoreFoundation+0x55554]  CFSetGetValue+0x24
C  [CoreFoundation+0x550b5]  __CFRunLoopFindMode+0xc5
C  [CoreFoundation+0x55889]  CFRunLoopAddSource+0xc9
C  [IOKit+0x3726f]  IOHIDDeviceScheduleWithRunLoop+0x66
C  [IOKit+0x3983a]  __IOHIDManagerDeviceApplier+0x1e8
C  [IOKit+0x3a7e2]  __IOHIDManagerDeviceAdded+0x153
C  [IOKit+0x39cf0]  __IOHIDManagerSetDeviceMatching+0xff
C  [jna4455466734030689331.tmp+0x1328]  **_hid_enumerate_**+0xc8
C  [jna8427230249487447742.tmp+0xe134]  ffi_call_unix64+0x4c
C  0x000070000df52dd8

And the one I'm experiencing it's inside the **_read_thread()_** when unplugging an active device. The  device's **_Source_** (IOKit code) inside the RunLoop encounters something NULL and blows up. My problem is how to influence this behavior at the hidapi layer, attempted to sync the RunLoopStop, changed from CFRunLoopRunInMode to Default, modified read_thread to basically only move the device to the thread's runLoop + register the report_callback + call and block with CFRunLoopRun + cleanup and return. And none of these options prevent the crash inside the IOKit routine execution of this device's source. I do believe they should have some protection to 'know' when the device was disconnected and don't attempt to process another report, right?