signal11/hidapi #339: hid_get_feature_report, where do yiu specifiy the report id?

ulao, 2017-06-17:
I'm really confused on this? I have a few feature reports that I need to get data from but I do not see where you specify the report id.  I tried using the first element in the array but I'm guessing this array is for the returned data. Does it go in the device somewhere? Surly it was designed to work with report other then a RID of 0?

 /// Return Type: int
        ///device: hid_device*
        ///data: unsigned char*
        ///length: size_t->unsigned int
        [DllImport(DLL_FILE_NAME, CallingConvention = CallingConvention.Cdecl)]
        public static extern int hid_get_feature_report(IntPtr device, [Out] byte[] buf_data, uint length);

signal11, 2017-06-19:
The report number is the first byte of the array.

ulao, 2017-06-19:
Are you sure, on a get? I know on a send that is the case. I tried putting data[0] = rpID on a get but no dice. So you are implying it does hits?

byte[0] =18;//report id dec .18
hid_get_feature_report(IntPtr device, [Out] byte[] buf_data, uint length);
return data will be {RID,returned data1, ,returned data1, ,returned data1, ....}

?

signal11, 2017-06-19:
Yes. Look at the code.

ulao, 2017-06-19:
Ill look back in to it thx.

closed 2017-06-19 by ulao