signal11/hidapi #340: hid_enumerate freeze

almindor, 2017-06-18:
hid_enumerate freezes for me sometime both on windows and linux. I'm using "native" usb-insert and remove detection (`WM_DEVICECHANGE` on windows and `libudev` on linux) and when a change is detecte call `hid_enumerate`
I also call enumerate right on app start to detect already-inserted device.

It seems that sometimes `hid_enumerate` just freezes. I'm not using a thread for the basic hidapi calls (just reads) so this freezes my whole app.

Is this behavior normal? My understanding was that `hid_enumerate` shouldn't block (worst case scenario it should just not list my device)

signal11, 2017-06-19:
Have you debugged into it?

almindor, 2017-06-19:
Not yet, was busy putting things together but I'll try tomorrow.

almindor, 2017-06-19:
Seems it goes all the way down to libusb:

```#0  0x00007ffff3e7f2bd in poll () from /usr/lib/libc.so.6
#1  0x00007ffff39846ec in ?? () from /usr/lib/libusb-1.0.so.0
#2  0x00007ffff398576a in libusb_handle_events_timeout_completed () from /usr/lib/libusb-1.0.so.0
#3  0x00007ffff39858b0 in libusb_handle_events_completed () from /usr/lib/libusb-1.0.so.0
#4  0x00007ffff39861d9 in ?? () from /usr/lib/libusb-1.0.so.0
#5  0x00007ffff3986539 in libusb_control_transfer () from /usr/lib/libusb-1.0.so.0
#6  0x00007ffff7bd75e8 in libusb_get_string_descriptor (length=64, data=<optimized out>, langid=0, 
    desc_index=0 '\000', dev_handle=0x8a9ad0) at /usr/include/libusb-1.0/libusb.h:1797
#7  is_language_supported (lang=0, dev=0x8a9ad0) at hid.c:364
#8  get_usb_string (dev=0x8a9ad0, idx=<optimized out>) at hid.c:411
#9  0x00007ffff7bd7a38 in hid_enumerate (vendor_id=21324, product_id=1) at hid.c:590
```

signal11, 2017-06-20:
Make a debug build of libusb and see if you can find what's failing and why. Also try strace.

almindor, 2017-06-20:
Here's what I got:

```
1   poll                                                                                                                                                                                                               0x7ffff3e7e2bd 
2   poll                                                                                                                                                                                    poll2.h               46   0x7ffff39836ec 
3   handle_events                                                                                                                                                                           io.c                  2151 0x7ffff39836ec 
4   libusb_handle_events_timeout_completed                                                                                                                                                  io.c                  2345 0x7ffff398476a 
5   libusb_handle_events_completed                                                                                                                                                          io.c                  2444 0x7ffff39848b0 
6   sync_transfer_wait_for_completion                                                                                                                                                       sync.c                51   0x7ffff39851d9 
7   libusb_control_transfer                                                                                                                                                                 sync.c                130  0x7ffff3985539 
8   libusb_get_string_descriptor                                                                                                                                                            libusb.h              1797 0x7ffff7bd6758 
9   is_language_supported                                                                                                                                                                   hid.c                 301  0x7ffff7bd6758 
10  get_usb_string                                                                                                                                                                          hid.c                 346  0x7ffff7bd6758 
11  hid_enumerate                                                                                                                                                                           ```

almindor, 2017-06-20:
I found out that it seems to happen when I do some transfer between the device and then kill the app which probably means the connection is not closed neatly.

Next time the app runs and calls enumerate it freezes here and I have to remove the device and re-insert.
If I close the app normally it works ok on next run. Also the kill is not performed "in the middle" of communicating, just when it's idle.