signal11/hidapi #345: multiple thread pick up cross talk?

ulao, 2017-07-03:
I'm not sure how this is even possible but was wondering if there is a solution here.  I have four threads running on 4 identical devices where the USB ID's are the only thing that is not the same. 

when I do   HidApi.hid_get_feature_report(DeviceHandle, buffer, (uint)length);
It just pics any of the USB traffic not the one that was created in the handle. 

I could add private ushort _VendorID, _ProductID; to my class and check it in the hid_get_feature_report response but I do not know how to get the ID from the handle? Can I cast it to a struck of some kind?

also Why is it picking up the cross talk? I though the HID feature requests send the PID/VID for verification?

signal11, 2017-07-10:
Which OS?

I've never seen anything like this before.

ulao, 2017-07-11:
windows 7 pro, ill try to get more detail.

ulao, 2017-07-13:
Double check and this is the case. Though you may never have seen this because most people will never have 4 devices the same name with changing IDs.  I wonder if there is a bug? Is there a way to get the ID from the hid_get_feature_report response, this could be an easy fix.