signal11/hidapi #348: Get/set feature report return Incorrect function on WIndows 8-10. On linux and mac work normally

zanpakutou22, 2017-07-14:
Hi,

I want to get HID signals from headsets. I built hidapi for linux, mac and windows. I run hidtest with my VID and PID. It is work fine on Linux and Mac platforms. I can get all control signals.  But on Windows Platforms (win 8.1, win 10) get/set feature report return Incorrect function and no data was read.

For testing I using Plantronics headsets.

How I can fix this issue?

signal11, 2017-07-14:
Can you debug into the code? Are you putting the report number as the first byte? Are you requesting the proper number of bytes (indicated by the report descriptor)? Windows is more strict about these things than the other OS's.

zanpakutou22, 2017-07-14:
I'm sorry, I was wrong. On Linux and Mac I also unable to get/set feature report, but I can read data. 

I more needs hid_read functionality.

Do you have any idea, why hid_read does not work?

If it helps, I can share code..but I only change VID and PID in your hidtest

signal11, 2017-07-14:
If all you're doing is running the test program your device probably doesn't have feature reports.

closed 2017-07-28 by zanpakutou22