signal11/hidapi #350: hid_read returns 0 on Windows Platforms, on Mac and Linux works fine

zanpakutou22, 2017-07-17:
Hi,

hid_read returns 0 on Windows Platforms:
In first ReadFile return 0, after that WaitForSingleObject always return 258. 
I could not get any data on Windows, but on Linux and Mac hid_read returned valid data.

For testing I using Plantronics headsets. I run hidtest with my VID and PID. hid_open works fine and func hid_set_nonblocking(handle, 1) was called.

How I can fix this issue? Do you have any idea?

signal11, 2017-07-17:
I don't have any specific idea. Are you using the latest from git (git clone ...) ?

zanpakutou22, 2017-07-17:
Yes, latest version.

It is strange, but I can get data signals from some Plantronics headsets, and can not get from Other Plantronics headsets.

For example: from Plantronics DA60 I can get signals, also I can get signals from any Logitech headsets
but I can not get any signals from Plantronics C210

tresf, 2017-07-17:
I have a `W01` and use a port of signal11 which was ported to Java.  We receiving the events from the `W01` base without any problems.

What you are describing sounds similar to a symptom we have if the device is already claimed (such as through the Plantronics spokes/hub software) which at the time of writing this is only available for Windows and Mac.

Is there a chance the `C210` is claimed by another application?

pinsonn, 2017-07-20:
I'm receiving reports of precisely the same issue while reading data from a custom hid device using latest release. 

TL/DR -- In latest update to Win10, after boot hid_read returns 0. Removing the device in device manager and scanning for changes to reattach device seems to fix the issue, and hid_read will function properly until power is secured. 

Some amplifying info:

- device enumerates fine (via hid_enumerate)
- file handle is successfully created and returned (via hid_open). File handle is created w/ shared read/write access, per baseline. 
- In Win10, after computer boots, hid_read returns 0. Hot plugging, restarting program, etc, do nothing to remedy the issue. Leaving device unattached during boot doesn't change the behavior. While hid_read returns 0, reading from device using a file handle created independent of hidapi works just fine. Phrased another way, while the program using hidapi is unable to read from the device, I can read from the device using a second program that does not use hidapi. 
- after removing the hid device in device manager and scanning for changes to reattach, file handle address is the same as before, but hid_read works fine (returns non-zero value of bytes read). Program can be restarted until the shutdown and hid_read continues to function like it should. 

I'm happy to explore further, but whatever is happening appears to be happening at a very low level in the OS; not really sure where to look next.

zanpakutou22, 2017-07-21:
I found that on my PC headset device have two path, but hidapi during opening device using the first path.
If I change code to use the second path, I can get all signals.

It is strange, but some of headsets have only one path, some have two and some have 3 path!!!!!!!!

[image]

signal11, 2017-07-24:
I've not noticed the "col" parameter. Can you find out what that means?

zanpakutou22, 2017-07-25:
It is number of HID collection
When you do HidD_GetPreparsedData in funct hid_enumerate, you can find CollectionNumber in _HIDP_COLLECTION_DESC, but I may wrong.

https://msdn.microsoft.com/library/windows/hardware/ff539679

typedef struct _HIDP_COLLECTION_DESC
{
   USAGE       UsagePage;
   USAGE       Usage;

   UCHAR       CollectionNumber;
   UCHAR       Reserved [15]; // Must be zero

   USHORT      InputLength;
   USHORT      OutputLength;
   USHORT      FeatureLength;
   USHORT      PreparsedDataLength;

   PHIDP_PREPARSED_DATA             PreparsedData;
} HIDP_COLLECTION_DESC, *PHIDP_COLLECTION_DESC;

closed 2017-07-28 by zanpakutou22

deltabeard, 2020-03-09:
Why was this issue closed? I'm currently experiencing this issue on Windows 10.

Youw, 2020-03-09:
please see #373