signal11/hidapi #353: Device freeze on hid_enumerate

heberallred, 2017-08-31:
I'm having the issue on Windows 10 as well as Ubuntu Linux.

When I call hid_enumerate, most of the time it works fine, but sometimes it will cause my mouse to freeze.  To fix the issue, I have to unplug and plug the mouse back in.  I'm experiencing this with my own custom device, as well as another gaming mouse that I purchased.

Any ideas why it would cause devices to freeze when simply doing an enumerate?

heberallred, 2017-08-31:
Here's a little more info:
In further testing, I found that if I specify a Vendor ID and Product ID, the enumerate call does not cause a freeze.  It freezes some devices when I pass "wildcards" 0 for Vendor ID and 0 for Product ID (which returns all devices).

jasonbrent, 2017-09-29:
My best guess is that you likely have some USB hardware that is misbehaving.