signal11/hidapi #356: hid_write return value Linux vs Windows

JasperNuytten, 2017-09-12:
Hi,

first off, great library, works like a charm!
Secondly, I'm opening this thread for knowledge purposes, I'm sorry if this is the wrong place.

I'm communicating with a custom-made USB HID device, it does not use any Report ID.
When using **Ubuntu** [16.04 LTS] using hidraw I'm able to write the following:
```cpp
uint8_t packetSizeInByte = 64;
unsigned char toSend[packetSizeInByte] = {0};
toSend[0] = 0x0b;
hid_write(hidHandler, toSend, packetSizeInByte)
```
This works like a charm, my device responds accordingly and hid_write returns the correct number of bytes, in this case 64. 
But I can also write this:
``` cpp
uint8_t packetSizeInByte = 64; 
unsigned char toSend[packetSizeInByte] = {0};
toSend[1] = 0x0b;
hid_write(hidHandler, toSend, packetSizeInByte)
```
Again, this works and the device responds as expected.

However, on **Windows 10** the first piece of code will not work, the device won't do anything. The second piece of code will work and the device responds accordingly.

Here are my questions:
1. On Windows,  hid_write returns 65 bytes instead of 64, although my device works with 64 byte packages. And it works anyway? Why/How? Is it because the used kernel driver adds an extra byte? If so, why? I thought I already stated that I'm not using any Report ID by leaving the first byte empty (or 0)?

2. From searching older issues and threads I understand this has something to do with Windows being stricter than Linux. How so?

I got the feeling I'm lacking some interesting knowledge here and was hoping somebody could help me out with my questions. If anything is unclear please let me know, I'll try to elaborate a little more.

Thanks in advance!

jasonbrent, 2017-09-29:
Look in hid.c at hid_write which is a wrapper for set_report(); if data[0] is 0, data has "1" added to it. In your second exmaple, toSend[0] would be 0 and match this condition.

`       if (data[0] == 0x0) {
                /* Not using numbered Reports.
                   Don't send the report number. */
                data_to_send = data+1;
                length_to_send = length-1;`

JasperNuytten, 2017-10-03:
Okay, that explains my first question a bit. But why does example 1 and 2 work on Ubuntu and only example 2 works on Windows?