signal11/hidapi #361: Can't Get Input / Output Feature Report Buffer Size in Win10

viclin920, 2017-10-27:
Hi all,

I'm working on Win10 64bit with USB2.0 HID (Consumer Devices).
I got a problem when I set output feature report buffer size over than 8,191 bytes in our device, the report from caps.OutputReportByteLength always shows 0
The API I use is as below shows:

HID_API_EXPORT hid_device * HID_API_CALL hid_open_path(const char *path)
{
         ...
	nt_res = HidP_GetCaps(pp_data, &caps);
	if (nt_res != HIDP_STATUS_SUCCESS) {
		register_error(dev, "HidP_GetCaps");	
		goto err_pp_data;
	}
    dev->output_report_length = caps.OutputReportByteLength;
    dev->input_report_length = caps.InputReportByteLength; 
   ...
}

Does anyone know what the problem is? 
Thanks.

ulao, 2018-01-31:
See if you have winusb as the driver instead of hidusb. You can use zadig to check.

viclin920, 2018-09-05:
Hi ulao,
Thank you for your reply.
I checked the hidusb driver is used by default.

any suggestion?