signal11/hidapi #367: hid_write: number of bytes written does not match report size?

pdged, 2017-11-28:
I have an output report of 64 bytes, and it is using report ids.
The number of bytes written is always 64, and not 65. Is this correct ?

```
const uint8_t report_id = 1;
const int report_size = 64;
const int buf_size = report_size + 1; // for the report id
uint8_t buf[buf_size];
buf[0]=report_id;
auto num_bytes_written = hid_write(device, buf, buf_size);
if (num_bytes_written != buf_size) {
// error
}
```

todbot, 2018-01-01:
Which OS are you on?  
I believe this is a problem currently with Windows `hid_write()`.
See issue #356 and #255.

ulao, 2018-01-31:
Make sure to remove the driver if on windows, Windows cashes the descriptor.  If you make changes to a size, you must remove it and let it reinstall.