signal11/hidapi #368: Segmentation fault with no context

Christopher154, 2017-12-04:
I am trying to connect my raspberry pi to a htc vive tracker and use HIDAPI to read and write information off it but when I compile and run the script all I get is segmentation fault without saying what line is causing the issue
```C
#include <stdio.h>
#include <stdlib.h>
#include <hidapi/hidapi.h>

int main (int argc, char* argv[])
{
        printf("This starts");
        int res;
        hid_device *handle;
        unsigned char dataA [6];
        unsigned char dataB [12];

        res = hid_init();
        printf("Working");
        handle = hid_open(0x28de, 0x2022, NULL);

        dataA[0] = 0xB3;
        dataA[1] = 0x04;
        dataA[2] = 0x03;
        dataA[3] = 0x00;
        dataA[4] = 0x00;
        dataA[5] = 0x00;

        dataB[0] = 0xB4;
        dataB[1] = 0x0A;
        dataB[2] = 0x00;
        dataB[3] = 0x01;
        dataB[4] = 0x00;
        dataB[5] = 0x00;
        dataB[6] = 0x00;
        dataB[7] = 0x00;
        dataB[8] = 0x00;
        dataB[9] = 0x00;
        dataB[10] = 0x00;
        dataB[11] = 0x00;

        hid_write(handle, dataA, 6);
        hid_write(handle, dataB, 12);

        res = hid_exit();
}
```

swt2c, 2017-12-10:
You are not checking the return values of `hid_init()`, `hid_open()` or `hid_write()`.  I would start there.  My bet is on hid_open returning NULL.

To debug a segmentation fault, you generally want to feed a core file into gdb or run your program under gdb and re-create the segfault.

dos-ise, 2018-01-31:
Having the same Problem on my rasberry.

dos-ise, 2018-02-01:
It seems that hid_open returns a zero Pointer on raspberry. So @swt2c was right in my case. But i don't know why because hid_enumerate returns the device.

signal11, 2018-02-01:
Use gdb and step into hid_open(). Is it a permissions issue?