signal11/hidapi #371: Undocumented read limitation

bobinmurphy, 2018-01-07:
There is, I believe, an undocumented limitation in the read function.  Some USB devices which claim to be HID capable can return enough data before the user can read them that the internal buffer chain for storing input Reports is over-run, and this is never reported to the user.  The hard coded limitation is set to 30 and when unread input Reports exceed this they are discarded without an error being posted to the user.  Suggest returning a buffer overrun error on read on a read timeout or when no more data is available if an overrun was previously seen.  It would also be nice if there was a user callable function that could change this limit, similar to the set_blocking call, or if there was an optional parameter on the open call.

dxon, 2018-01-16:
Hello. It looks very similar to my situation. I have device which gets messages from laptop via USB and then reads and writes it to Can bus. I have a software written on Qt.

So I found that only 16203 messages could be read. And I get some system error after this in release version of software or SEGFAULT msg in Qt debbuger when trying to cach the reason of this error.

Then I tried to work with this USB2CAN device and wrote very similar software on C# in Visual Studio using other API for USB (https://github.com/brunsa2/Senior-design-USB-interface). And I don't have such issue there.

Now I'm gonna try to use Qt and libusb API to make sure that is not issue on Qt side. I'll post my result here later.

bobinmurphy, 2018-01-16:
I was able to solve the problem in my app by increasing the number of unread Input Reports in the HIDAPI library from 30 to 200 and implementing a read loop in my code that runs out of a timer callback.  The timer is firing off every 250ms and it drains the HID input chain into a local buffer for later processing.  My USB device probably isn't the best design for HID as it only returns a single data byte on each Input Report and it's largest message is about 150 bytes total.  Messages themselves are only sent in from the device every 2 or 3 seconds though.  Had it packed up data into 8-byte length Input Reports it would have worked fine with an unmodified version of the HIDAPI library.

If the Qt debugger is built on GDB and you have more than about 350K of local data you might be bumping into an undocumented limitation of GDB.  With too much local data GDB can't access it and will throw SEGFAULTs whenever your program accesses a data element that's outside GDB's limits.  Haven't figured out how to work around this one myself.

dxon, 2018-01-16:
Hello. Thanks for reply. Could you share your code please?

Btw, my app crashes even if it built as static and runs without any debugger.

bobinmurphy, 2018-01-16:
The line(s) I changed in the HIDAPI library were the comment on line 763 and the constant on line 766.  I replaced the "30" in both places with a "200" and rebuilt the library for use with my app.

Dixon, if you'd like copies of the timer code I used where I unload the HIDAPI Input Report buffer chain, could you send me an e-mail to rnestor@tx.rr.com?  I don't think that timer code is of much use to the rest of this forum.

dxon, 2018-01-17:
Hello dear bobinmurphy. I've send you message to e-mail. Check it please.

dxon, 2018-01-17:
Hello dear Bob. Got your letter with code. Thanks a lot. 
So the same story with libusb-1.2. Plus I see memory leaking while software reads and writes messages to USB2CAN device. Have a little progress here - now I get SEGFAULT after 28745 messages.

And my similar software on C# works more stable (I would say - it works perfect without any bug or issue).

What is wrong with Qt/libusb? I don't have any idea... :-(

bobinmurphy, 2018-01-17:
The app I'm working with was ported from C# where the original author said it was working fine as well.  I found some logic errors in the code converting it to C, but once converted it ran fine for years on a BSD system without libusb or hidapi.  In porting it to Linux I decided to use libusb and hidapi as well as converting much of it to C++, and now I'm chasing SEGFAULTs too. Some are mine, no doubt, but some appear to be coming from how threads and forked processes inherit or access variables from their parent process.   

I'm not all that familiar with C# and how it does memory management, but I'd venture a guess that the problems you're seeing are probably related to undefined/uninitialized variables.  Some systems and languages initialize newly allocated memory to zero while others don't and porting a working program between these two environments can generate a lot of unexpected behavior as a result.