signal11/hidapi #376: winusb

ulao, 2018-01-31:
As I gather it, winusb sort of cripples the control transfers. I don't know much about this but read some about it. It would seem that win10 or one of the popular installs of win10 is putting winusb on the system and replacing hidusb. 

Does anyone know why winusb is not working or somehow manages to interfere with control  transfers. It is something we can fix?

signal11, 2018-02-01:
Is winusb now able to communicate with HID devices? Last I was aware, it was not.

ulao, 2018-02-01:
I wondering if I choose the option to build the dll in to the application if this will fix it?

signal11, 2018-02-01:
winusb, as far as I know, will not open HID devices, even though HID devices are USB. It blocks the opening of HID devices.

This was the case last I looked.

ulao, 2018-02-01:
I wonder why winusb is causes issue with this project then? It definitely  opens the device but communicate is all bugged. Maybe hidapp.dll is opening it but winusb is communicating? So your project has a option to build whit hiapi.dll, ill try that.

mcuee, 2018-04-08:
If your application uses HIDAPI and the user's device is not using the HID driver (not HID device), then of course your application will not work. How can you open the device if the driver is WinUSB?