signal11/hidapi #377: embed HIDAPI directly MVS

ulao, 2018-02-01:
How is this done in MVS? I currently must have the hidapi.dll in my execute folder. How can I built it directly in to my application. Idea being to avoid winusb.  winusb is interfering in some way and I'm hoping this will fix that.

ulao, 2018-04-08:
MVC = Microsoft visual studio. 
Winusb on window 10 some how gets attached to a hid driver. I do not understand this at all, nor do I understand how it occurs. All I know is when you look at the install drivers with a program called Zadig you can see what driver has hidusb and what has winusb.  Whenever its winusb the feature report just do not work right. The hope here was that embedding it would force it.  Though as it turns out zadig allow one to switch and thus fixing the issue.  So this topic can be left alone if unless anyone can provide details on how to embed with visual studio, that is still helpful even though it does not address my concern.

mcuee, 2018-04-08:
Maybe you can name a few device behave this way. Are they USB composite device (HID + WinUSB)? If it is WinUSB, then you can not use HIDAPI.

As you mentioned, Zadig can switch driver. Technically you can use WinUSB driver for HID device but then it will lose the function as an HID device and not able to use HIDAPI.

mcuee, 2018-04-08:
To use with visual studio, you can add the header file and the library to your project.

ulao, 2018-04-08:
"As you mentioned, Zadig can switch driver. Technically you can use WinUSB driver for HID device but then it will lose the function as an HID device and not able to use HIDAPI." - Yes that is what I see. In my case this is a HID device not winusb. I'm not sure why win10 is using winusb with it but it does not in all cases.

mcuee, 2018-04-08:
What is this device? Can you print out the USB descriptor?

> --
Xiaofan

ulao, 2018-04-08:
Just a game pad with a huge force feedback descriptor almost identical to the ms side winder.

mcuee, 2018-04-11:
I believe Windows will not use WinUSB for an HID device unless user switch the driver to WinUSB.

There is also a possibility that the device has two interfaces, one is HID and use the HID driver. The other is a custom interface using WinUSB.