signal11/hidapi #379: Device paths in macOS

mhavu, 2018-02-07:
I have a type of USB device that is not a HID controller but advertises itself as one, so I'm using HIDAPI to communicate with it. A number of these devices may be plugged in and out at any time. Polling them with `hid_enumerate()` and `hid_read()` is inefficient and cumbersome in this case, so I would rather move to hotplug event notifications. I started with a macOS implementation, but I plan to add this for other platforms as well.

I managed to register callbacks with `IOServiceAddMatchingNotification()`. In my deviceAdded callback, I first get the device path with `IORegistryEntryGetPath()`, and pass it to `hid_open_path()`. However, the `IOHIDDeviceCreate()` call returns NULL, and opening the device fails. This seems to be because the path I get from `IORegistryEntryGetPath()` is not the same as the path I get from `hid_enumerate()` (see below). How can I get paths that are compatible with HIDAPI? Should I be using `IOHIDManagerRegisterDeviceMatchingCallback()` and
`IOHIDManagerRegisterDeviceRemovalCallback()` instead of `IOServiceAddMatchingNotification()`? Will that interfere with `hid_device_removal_callback`?

The paths I get from `IORegistryEntryGetPath()` are of the form
_IOService:/IOResources/AppleUSBHostResources/AppleUSBLegacyRoot/AppleUSBXHCI@14000000/FOO@14610000_,
whereas the paths I get from `hid_enumerate()` are of the form
_IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS09@14600000/USB2.0 Hub@14600000/AppleUSB20Hub@14600000/AppleUSB20HubPort@14610000/FOO@14610000/IOUSBHostInterface@0/IOUSBHostHIDDevice@14610000,0_.

mhavu, 2018-02-07:
Seems like using `IOHIDManagerRegisterDeviceMatchingCallback()` and `IOHIDDeviceGetService()` is the way to go. I managed to get a small test app working, but when I add the functions to my real application, the callbacks are never called. Will have to investigate.

mhavu, 2018-02-07:
Hmm... maybe I should've gone the other way, and modified HIDAPI to use IOService instead: [https://bugs.chromium.org/p/chromium/issues/detail?id=413978](url)

jdk, 2018-02-14:
Can you take a look at this to see if it works for you: https://github.com/flirc/hidapi/commit/8d251c3854c3b1877509ab07a623dafc8e803db5

I'm working with Felix on merging this into the main branch.

mhavu, 2018-02-16:
@jdk Thanks, I will have to get back to you on this one. I got my application to the point where everything seems to work fine (at least on macOS 10.13) using HID Manager. You need to use a separate thread and call `CFRunLoopRun()` in that thread to get it working. However, our testers reported that the devices are not detected in Yosemite. I will need to investigate this. If IOService is in fact more lightweight, and support for Yosemite is restored, I will go with your version.

closed 2018-02-16 by mhavu

mhavu, 2018-04-04:
Using IOService restored the support for Yosemite, so I tried your code. It is not ideal, since the entry ID (unlike the path) changes each time the device is disconnected and reconnected. I think I can live with unnecessarily reinstantiating a few objects, though.

jdk, 2018-04-04:
@mhavu Yes, you'd need to reenumerate a list again and find the device. A bit taxing but nonetheless, worth it considering this is a bug in OSX and prevents you from opening the device by path at all.

mhavu, 2018-04-05:
In this case finding the device is impossible, since the manufacturer hasn't bothered giving the devices unique serial numbers. The only way to identify the device is to read it, and if I can't open it, well...

I guess my options are 1) keep using HID Manager, and lose Yosemite support, or 2) rewrite HIDAPI to use the lighter-weight IOService calls instead of HID Manager. I don't know what issues I will run into, if I choose option 2.

mhavu, 2018-04-09:
I figured out a third option: Do not use the information about added or removed devices in the matching or removal callback at all. Instead, let HIDAPI enumerate the devices every time a matching or removal callback is called. This is sort of like polling, but substituting the callbacks for timer as the trigger. @jdk: Reading your previous comment again, I see this might be what you suggested. Like you said, a bit taxing, but still much better than polling.

jdk, 2018-04-10:
@mhavu  yup, sounds good to me.