signal11/hidapi #387: Android Won't Compile Detail error below

Rathna-K, 2018-03-25:
when I compile the Android.MK inside jni using NDK under windows, I am getting this error:
`Android NDK: APP_PLATFORM not set. Defaulting to minimum supported version android-14.
D:/adt-bundle-windows-x86_64-20140702/android-ndk-r16b/build//../build/core/build-binary.mk:693: Android NDK: Module hidapi depends on undefined modules: usb1.0
D:/adt-bundle-windows-x86_64-20140702/android-ndk-r16b/build//../build/core/build-binary.mk:693: Android NDK: Module hidapi depends on undefined modules: usb1.0
D:/adt-bundle-windows-x86_64-20140702/android-ndk-r16b/build//../build/core/build-binary.mk:693: Android NDK: Module hidapi depends on undefined modules: usb1.0
D:/adt-bundle-windows-x86_64-20140702/android-ndk-r16b/build//../build/core/build-binary.mk:693: Android NDK: Module hidapi depends on undefined modules: usb1.0
[arm64-v8a] Compile        : hidapi <= hid.c
C:/Users/Rathna/Downloads/hidapi-master/android/jni/../../libusb/hid.c:47:10: fatal error: 'libusb.h' file not found
#include <libusb.h>
         ^~~~~~~~~~
1 error generated.
make: *** [C:/Users/Rathna/Downloads/hidapi-master/android/obj/local/arm64-v8a/objs/hidapi/__/__/libusb/hid.o] Error 1
`

Alcander-Liu, 2018-08-04:
Any update of this issue? I've been facing the same problem.

cnlohr, 2019-08-23:
I will admit I am also stumped as to where to put the libusb1.0 files that had been built using it's JNI system.  I wonder if anyone will ever add a README to the android folder

malbit, 2019-08-26:
> I will admit I am also stumped as to where to put the libusb1.0 files that had been built using it's JNI system. I wonder if anyone will ever add a README to the android folder

You can put wherever you want it. Just add LIBDIR to project and that will sort things out

malbit, 2019-08-26:
Error above says there is ANDROID_NDK not set. Download NDK end export path to it and it should work. I am using Ubuntu Docker to build android libs and never had problems with it