signal11/hidapi #402: Spreads/HIDAPI

eriwm, 2018-07-10:
Hello, I dont think this is specifically a HIDAPI problem but I am trying to use HIDTRIGGER in spreads which needs HIDAPI to function as follows.
Trigger plugin that waits for a key press on any USB HID device. """
from __future__ import unicode_literals
import hidapi
import logging
import threading
import time
from spreads.plugin import HookPlugin, TriggerHooksMixin
from spreads.util import DeviceException
However when I try to configure Spreads and choose the HIDTRIGGER plugin I get the error -
ExtensionException: Missing Python dependency for extension 'hidtrigger':hidapi.
I have tried several ways of installing HIDAPI including downloading and building from the git repository.
Doing a file search shows that HIDAPI is installed but it seems to me that HIDTRIGGER cannot find HIDAPI so I don't think I am installing it in the right place.
Can someone please point me in the right direction. Thanks Eric.

todbot, 2018-07-10:
This sounds like a Python library question, not hidapi (a C library). 
I think instead you should be looking at: https://pypi.org/project/hidapi/
This is the Python hidapi wrapper that I often use.
You can install it with `pip install hidapi`.

eriwm, 2018-07-10:
Hello Tod,
Thank you for getting back to me so quickly. I have tried installing as 
you described but still get the same error message. I am using ubuntu 
16.04 and I think maybe the problem lies somewhere in the symlinks, 
guess I will just have to keep digging. I am not an expert so going to 
take some time.
Once again thanks.
Eric

todbot, 2018-07-10:
Looking at the setup file for `spreads`, I think you should instead be installing [hidapi-cffi](https://pypi.org/project/hidapi-cffi/).  But this should have gotten installed for you if you installed `spreads` via the normal means, as described here: https://pypi.org/project/spreads/

eriwm, 2018-07-10:
Thank you so much - installing hidapi-cffi seems to have fixed the 
problem. The version of Spreads I am using was downloaded from

https://github.com/DIYBookScanner/spreads.git I was never able to get the version from https://pypi.org/project/spreads/ to work properly.
You have saved me hours of work- thanks again.
Eric