signal11/hidapi #403: Windows static library

robotrovsky, 2018-07-31:
Hi,
i would like to build a static library for windows, but when i set the library from dynamic to static i get the following unresolved symbols when building hidtest:

```
1>hidapi.lib(hid.obj) : error LNK2001: Nicht aufgelöstes externes Symbol "__imp_SetupDiDestroyDeviceInfoList".
1>hidapi.lib(hid.obj) : error LNK2001: Nicht aufgelöstes externes Symbol "__imp_SetupDiEnumDeviceInterfaces".
1>hidapi.lib(hid.obj) : error LNK2001: Nicht aufgelöstes externes Symbol "__imp_SetupDiGetClassDevsA".
1>hidapi.lib(hid.obj) : error LNK2001: Nicht aufgelöstes externes Symbol "__imp_SetupDiGetDeviceInterfaceDetailA".
1>hidapi.lib(hid.obj) : error LNK2001: Nicht aufgelöstes externes Symbol "__imp_SetupDiEnumDeviceInfo".
1>hidapi.lib(hid.obj) : error LNK2001: Nicht aufgelöstes externes Symbol "__imp_SetupDiGetDeviceRegistryPropertyA".
```
What else am i missing??

Youw, 2018-07-31:
Add `Setupapi.lib` to your link libraries.

robotrovsky, 2018-07-31:
Thank you very much for the super fast help!

closed 2018-07-31 by robotrovsky