signal11/hidapi #413: How can I use this?

Goldname98, 2018-11-13:
I'm sorry, but I can't find any documentation showing how I can use this to read input data from a bluetooth device.

nonchip, 2018-11-24:
try reading the readme.

buaabyl, 2018-11-27:
Is there any compile or link error log?

nonchip, 2018-11-27:
@buaabyl well the first compile error seemed to be "using g++" instead of running make ;)

buaabyl, 2018-11-27:
the core library `hid.c` is c file, but `hidtest.cpp` is c++ file, you may using g++ to compile them all. or gcc for `hid.c` and `g++` for `hidtest.cpp`, and link them together.

nonchip, 2018-11-27:
true, you can, but it's not really how you should link this lib according to the readme :P

so probably better trying the makefile to prevent "layer 8" issues, and of course pasting the log on error, as you already said.

Goldname98, 2018-12-19:
Yes, I was trying to compile them all, but with some linker issues. It's ok, I found the hidtest file which is very helpful, thank you.

closed 2018-12-19 by Goldname98