signal11/hidapi #415: ATMEL-ICE: testgui works with libusb but not with hidraw

vasily-kartashov, 2018-12-07:
What can be the reason for `hidapi-hidraw-testgui` not seeing ATMEL-ICE device, while `hidapi-libusb-testgui` is able to connect to it just fine?

Also, when I remove the following udev ruleset

```
SUBSYSTEM!="usb", ACTION!="add", GOTO="avrisp_end"

ATTR{idVendor}=="03eb", ATTR{idProduct}=="2141", MODE="660", GROUP="dialout"

LABEL="avrisp_end"
```

both testgui tools see the device, but neither can connect to it.

vasily-kartashov, 2018-12-07:
Forgot to mention, this was on latests Fedora. When I run TestGUI On Mac OS X the device is not recognized either.

jonathancross, 2019-07-11:
This repo is no longer maintained, see #431 

You might try the [libusb mailing list](https://sourceforge.net/projects/libusb/lists/libusb-devel) or opening an [issue at libusb/hidapi](https://github.com/libusb/hidapi/issues).

Cheers!