signal11/hidapi #417: ./bootstrap: Can't find "LT_INIT"

thegergo02, 2019-01-03:
After running ./bootstrap, I get the following output:
```
$ ./bootstrap
+ autoreconf --install --verbose --force
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
autoreconf: configure.ac: tracing
autoreconf: configure.ac: not using Libtool
autoreconf: running: /usr/bin/autoconf --force
autoreconf: running: /usr/bin/autoheader --force
autoreconf: running: automake --add-missing --copy --force-missing
libusb/Makefile.am:11: error: Libtool library used but 'LIBTOOL' is undefined
libusb/Makefile.am:11:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
libusb/Makefile.am:11:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
libusb/Makefile.am:11:   If 'LT_INIT' is in 'configure.ac', make sure
libusb/Makefile.am:11:   its definition is in aclocal's search path.
libusb/Makefile.am:18: error: Libtool library used but 'LIBTOOL' is undefined
libusb/Makefile.am:18:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
libusb/Makefile.am:18:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
libusb/Makefile.am:18:   If 'LT_INIT' is in 'configure.ac', make sure
libusb/Makefile.am:18:   its definition is in aclocal's search path.
libusb/Makefile.am:4: error: Libtool library used but 'LIBTOOL' is undefined
libusb/Makefile.am:4:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
libusb/Makefile.am:4:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
libusb/Makefile.am:4:   If 'LT_INIT' is in 'configure.ac', make sure
libusb/Makefile.am:4:   its definition is in aclocal's search path.
linux/Makefile.am:1: error: Libtool library used but 'LIBTOOL' is undefined
linux/Makefile.am:1:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
linux/Makefile.am:1:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
linux/Makefile.am:1:   If 'LT_INIT' is in 'configure.ac', make sure
linux/Makefile.am:1:   its definition is in aclocal's search path.
mac/Makefile.am:1: error: Libtool library used but 'LIBTOOL' is undefined
mac/Makefile.am:1:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
mac/Makefile.am:1:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
mac/Makefile.am:1:   If 'LT_INIT' is in 'configure.ac', make sure
mac/Makefile.am:1:   its definition is in aclocal's search path.
windows/Makefile.am:1: error: Libtool library used but 'LIBTOOL' is undefined
windows/Makefile.am:1:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
windows/Makefile.am:1:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
windows/Makefile.am:1:   If 'LT_INIT' is in 'configure.ac', make sure
windows/Makefile.am:1:   its definition is in aclocal's search path.
autoreconf: automake failed with exit status: 1
thegergo02@thegergo02:~/Documents/hidapi$ autoreconf -f -i
libusb/Makefile.am:11: error: Libtool library used but 'LIBTOOL' is undefined
libusb/Makefile.am:11:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
libusb/Makefile.am:11:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
libusb/Makefile.am:11:   If 'LT_INIT' is in 'configure.ac', make sure
libusb/Makefile.am:11:   its definition is in aclocal's search path.
libusb/Makefile.am:18: error: Libtool library used but 'LIBTOOL' is undefined
libusb/Makefile.am:18:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
libusb/Makefile.am:18:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
libusb/Makefile.am:18:   If 'LT_INIT' is in 'configure.ac', make sure
libusb/Makefile.am:18:   its definition is in aclocal's search path.
libusb/Makefile.am:4: error: Libtool library used but 'LIBTOOL' is undefined
libusb/Makefile.am:4:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
libusb/Makefile.am:4:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
libusb/Makefile.am:4:   If 'LT_INIT' is in 'configure.ac', make sure
libusb/Makefile.am:4:   its definition is in aclocal's search path.
linux/Makefile.am:1: error: Libtool library used but 'LIBTOOL' is undefined
linux/Makefile.am:1:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
linux/Makefile.am:1:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
linux/Makefile.am:1:   If 'LT_INIT' is in 'configure.ac', make sure
linux/Makefile.am:1:   its definition is in aclocal's search path.
mac/Makefile.am:1: error: Libtool library used but 'LIBTOOL' is undefined
mac/Makefile.am:1:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
mac/Makefile.am:1:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
mac/Makefile.am:1:   If 'LT_INIT' is in 'configure.ac', make sure
mac/Makefile.am:1:   its definition is in aclocal's search path.
windows/Makefile.am:1: error: Libtool library used but 'LIBTOOL' is undefined
windows/Makefile.am:1:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
windows/Makefile.am:1:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
windows/Makefile.am:1:   If 'LT_INIT' is in 'configure.ac', make sure
windows/Makefile.am:1:   its definition is in aclocal's search path.
autoreconf: automake failed with exit status: 1
```

thegergo02, 2019-01-03:
Fixed it, I had to get the libtools-dev package.

jonathancross, 2019-07-11:
If this is something that can be improved with documentation, please add a PR here in the maintained fork:
https://github.com/libusb/hidapi

If it was just a mistake, please consider closing this issue. :-)

Cheers!

closed 2019-09-28 by thegergo02