signal11/hidapi #419: Not able to read/write on Windows 10

sd1234567, 2019-01-09:
I'm trying to establish communication between BLE HID device and Windows machine. I'm using custom HID descriptor. I could verify that the firmware on hardware is working using readymade api running on Windows & Mac. Now I'm trying to develop my own hidapi with the help of signal11 code. Byte is getting written but I cant see that on BLE device. So I cant verify whether its truely sending the byte. Also I'm trying to send byte from BLE device but I just get 'waiting' on the console. Please check image below - 
[image]

connerMa, 2022-02-08:
I have the same problem you have

Youw, 2022-02-08:
Please see #468