signal11/hidapi #423: Invalid "desired_access" in "open_device" at enumeration

vladimir-sk, 2019-03-22:
Only for Windows OS.

We should use "desired_access = GENERIC_READ" at the enumeration devices in the "open_device"-function.
So a valid code for the "open_device"-function:

static HANDLE open_device(const char *path, BOOL enumerate)
{
    HANDLE handle;
    DWORD desired_access = (enumerate)? GENERIC_READ : (GENERIC_WRITE | GENERIC_READ);
    DWORD share_mode = FILE_SHARE_READ|FILE_SHARE_WRITE;

    handle = CreateFileA(path,
        desired_access,
        share_mode,
        NULL,
        OPEN_EXISTING,
        FILE_FLAG_OVERLAPPED,/*FILE_ATTRIBUTE_NORMAL,*/
        0);

    return handle;
}

I have some USB HID devices that do not support the ability to send data and read data.
The “CreateFile” function always fails (for these devices) when I try to use a non-empty “desired_access” (GENERIC_READ, GENERIC_WRITE, GENERIC_WRITE | GENERIC_READ, ...).
But “CreateFile” return a "valid" handle (for these devices) when I use a empty “desired_access” (desired_access = 0) and "ReadFile"/"WriteFile" WInAPI functions always fails for this "valid" handle.

CaffeMrDe, 2019-06-25:
I also meet the question . I solve the questsion after I browse the address https://stackoverflow.com/questions/37213634/createfilea-fails-to-open-hid-device-in-windows . After then, I am meeting the question about read buffer, which show code -1 that hid_read(handle, buf, sizeof(buf)) return .

If you are continuing to notice this topic ,  we can discuss togethor.

CaffeMrDe, 2019-06-25:
with CreateFileA's params, e.g. replacing GENERIC_READ | GENERIC_WRITE with STANDARD_RIGHTS_READ | STANDARD_RIGHTS_WRITE. But I can't knwo why.....