signal11/hidapi #424: build problem on FreeBSD

cederom, 2019-04-28:
Hello world, I have build problem on FreeBSD.. any hints welcome :-)

```
(venv37bsd) pip install hidapi
Collecting hidapi
  Using cached https://files.pythonhosted.org/packages/c1/86/89df0e8890f96eeb5fb68d4ccb14cb38e2c2d2cfd7601ba972206acd9015/hidapi-0.7.99.post21.tar.gz
Requirement already satisfied: setuptools>=19.0 in /ztuff/cederom/(..)/venv37bsd/lib/python3.7/site-packages (from hidapi) (40.8.0)
Building wheels for collected packages: hidapi
  Building wheel for hidapi (setup.py) ... error
  ERROR: Complete output from command /ztuff/cederom/(..)/venv37bsd/bin/python3.7 -u -c 'import setuptools, tokenize;__file__='"'"'/tmp/pip-install-54v1qpmp/hidapi/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-0v4ql11i --python-tag cp37:
  ERROR: running bdist_wheel
  running build
  running build_ext
  cythoning hid.pyx to hid.c
  /tmp/pip-install-54v1qpmp/hidapi/.eggs/Cython-0.29.7-py3.7-freebsd-12.0-RELEASE-p3-amd64.egg/Cython/Compiler/Main.py:367: FutureWarning: Cython directive 'language_level' not set, using 2 for now (Py2). This will change in a later release! File: /tmp/pip-install-54v1qpmp/hidapi/hid.pyx
    tree = Parsing.p_module(s, pxd, full_module_name)
  building 'hid' extension
  creating build
  creating build/temp.freebsd-12.0-RELEASE-p3-amd64-3.7
  creating build/temp.freebsd-12.0-RELEASE-p3-amd64-3.7/hidapi
  creating build/temp.freebsd-12.0-RELEASE-p3-amd64-3.7/hidapi/libusb
  cc -pthread -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -O2 -pipe -fstack-protector -fno-strict-aliasing -fPIC -Ihidapi/hidapi -I/usr/include/libusb-1.0 -I/ztuff/cederom/(..)/venv37bsd/include -I/usr/local/include/python3.7m -c hid.c -o build/temp.freebsd-12.0-RELEASE-p3-amd64-3.7/hid.o
  cc -pthread -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -O2 -pipe -fstack-protector -fno-strict-aliasing -fPIC -Ihidapi/hidapi -I/usr/include/libusb-1.0 -I/ztuff/cederom/(..)/venv37bsd/include -I/usr/local/include/python3.7m -c hidapi/libusb/hid.c -o build/temp.freebsd-12.0-RELEASE-p3-amd64-3.7/hidapi/libusb/hid.o
  hidapi/libusb/hid.c:458:18: warning: passing 'const char **' to parameter of type 'char **' discards qualifiers in nested pointer types [-Wincompatible-pointer-types-discards-qualifiers]
          res = iconv(ic, &inptr, &inbytes, &outptr, &outbytes);
                          ^~~~~~
  /usr/include/iconv.h:58:41: note: passing argument to parameter here
  size_t  iconv(iconv_t, char ** __restrict,
                                           ^
  1 warning generated.
  creating build/lib.freebsd-12.0-RELEASE-p3-amd64-3.7
  cc -pthread -shared -L/usr/local/lib -fstack-protector build/temp.freebsd-12.0-RELEASE-p3-amd64-3.7/hid.o build/temp.freebsd-12.0-RELEASE-p3-amd64-3.7/hidapi/libusb/hid.o -L/usr/local/lib -lusb-1.0 -lpython3.7m -o build/lib.freebsd-12.0-RELEASE-p3-amd64-3.7/hid.so
  /usr/bin/ld: error: unable to find library -lusb-1.0
  cc: error: linker command failed with exit code 1 (use -v to see invocation)
  error: command 'cc' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for hidapi
  Running setup.py clean for hidapi
Failed to build hidapi
Installing collected packages: hidapi
  Running setup.py install for hidapi ... error
    ERROR: Complete output from command /ztuff/cederom/(..)/venv37bsd/bin/python3.7 -u -c 'import setuptools, tokenize;__file__='"'"'/tmp/pip-install-54v1qpmp/hidapi/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-l7356dz0/install-record.txt --single-version-externally-managed --compile --install-headers /ztuff/cederom/(..)/venv37bsd/include/site/python3.7/hidapi:
    ERROR: running install
    running build
    running build_ext
    skipping 'hid.c' Cython extension (up-to-date)
    building 'hid' extension
    creating build
    creating build/temp.freebsd-12.0-RELEASE-p3-amd64-3.7
    creating build/temp.freebsd-12.0-RELEASE-p3-amd64-3.7/hidapi
    creating build/temp.freebsd-12.0-RELEASE-p3-amd64-3.7/hidapi/libusb
    cc -pthread -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -O2 -pipe -fstack-protector -fno-strict-aliasing -fPIC -Ihidapi/hidapi -I/usr/include/libusb-1.0 -I/ztuff/cederom/(..)/venv37bsd/include -I/usr/local/include/python3.7m -c hid.c -o build/temp.freebsd-12.0-RELEASE-p3-amd64-3.7/hid.o
    cc -pthread -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -O2 -pipe -fstack-protector -fno-strict-aliasing -fPIC -Ihidapi/hidapi -I/usr/include/libusb-1.0 -I/ztuff/cederom/(..)/venv37bsd/include -I/usr/local/include/python3.7m -c hidapi/libusb/hid.c -o build/temp.freebsd-12.0-RELEASE-p3-amd64-3.7/hidapi/libusb/hid.o
    hidapi/libusb/hid.c:458:18: warning: passing 'const char **' to parameter of type 'char **' discards qualifiers in nested pointer types [-Wincompatible-pointer-types-discards-qualifiers]
            res = iconv(ic, &inptr, &inbytes, &outptr, &outbytes);
                            ^~~~~~
    /usr/include/iconv.h:58:41: note: passing argument to parameter here
    size_t  iconv(iconv_t, char ** __restrict,
                                             ^
    1 warning generated.
    creating build/lib.freebsd-12.0-RELEASE-p3-amd64-3.7
    cc -pthread -shared -L/usr/local/lib -fstack-protector build/temp.freebsd-12.0-RELEASE-p3-amd64-3.7/hid.o build/temp.freebsd-12.0-RELEASE-p3-amd64-3.7/hidapi/libusb/hid.o -L/usr/local/lib -lusb-1.0 -lpython3.7m -o build/lib.freebsd-12.0-RELEASE-p3-amd64-3.7/hid.so
    /usr/bin/ld: error: unable to find library -lusb-1.0
    cc: error: linker command failed with exit code 1 (use -v to see invocation)
    error: command 'cc' failed with exit status 1
    ----------------------------------------
ERROR: Command "/ztuff/cederom/(..)/venv37bsd/bin/python3.7 -u -c 'import setuptools, tokenize;__file__='"'"'/tmp/pip-install-54v1qpmp/hidapi/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-l7356dz0/install-record.txt --single-version-externally-managed --compile --install-headers /ztuff/cederom/(..)/venv37bsd/include/site/python3.7/hidapi" failed with error code 1 in /tmp/pip-install-54v1qpmp/hidapi/
(venv37bsd)

(venv37bsd) python --version
Python 3.7.3

(venv37bsd) uname -a
FreeBSD 0xCFC2 12.0-RELEASE-p3 FreeBSD 12.0-RELEASE-p3 GENERIC  amd64
```

jonathancross, 2019-07-11:
This repo is no longer maintained, see #431 

You might try the [libusb mailing list](https://sourceforge.net/projects/libusb/lists/libusb-devel) or opening an [issue at libusb/hidapi](https://github.com/libusb/hidapi/issues).

Cheers!

cederom, 2019-10-27:
Thank you @jonathancross :-)

closed 2019-10-27 by cederom