signal11/hidapi #425: macOS: location of IOKit.framework

0xDP, 2019-05-04:
Hi,

according to

<https://developer.apple.com/library/archive/documentation/MacOSX/Conceptual/OSX_Technology_Overview/SystemFrameworks/SystemFrameworks.html>

IOKit.framework is located at "/System/Library/Frameworks/", but hidapi/mac/hid.c:hidapi_IOHIDDeviceGetService() tries to find it at "/System/Library/". That fails and at least on macOS 10.14 the fallback method seems to not working too, so that hid_enumerate() cannot determine a path. A subsequent call of hid_open() fails due to the empty path. The following patch fixes this issue for me:

```
--- a/hidapi/mac/hid.c
+++ b/hidapi/mac/hid.c
@@ -303,7 +303,7 @@ static io_service_t hidapi_IOHIDDeviceGe
 	 * and the fallback method will be used.
 	 */
 	if (iokit_framework == NULL) {
-		iokit_framework = dlopen("/System/Library/IOKit.framework/IOKit", RTLD_LAZY);
+		iokit_framework = dlopen("/System/Library/Frameworks/IOKit.framework/IOKit", RTLD_LAZY);
 
 		if (iokit_framework != NULL)
 			dynamic_IOHIDDeviceGetService = dlsym(iokit_framework, "IOHIDDeviceGetService");
```

Regards,
Dennis

0xDP, 2019-05-04:
Sorry, that relates to cython-hidapi.

0xDP, 2019-05-04:
Reopened, because this have to be fixed here (upstream) instead of in cython-hidapi.

mrpippy, 2019-05-05:
Wow, I guess this is just a typo I made years ago but it somehow hasn’t caused any problems until now. Can you open a pull request with this change?

jonathancross, 2019-07-11:
Hi @0xDP and @mrpippy  -- is there any way we can get you all to move this over to the new actively maintained fork of this project at https://github.com/libusb/hidapi ?  Thanks!