signal11/hidapi #428: ZERO bytes not read

dekosan, 2019-05-29:
Hi,
I'm using the hidapi on an FTDI FT260 on MacOS-Xcode-Obj-C. It works fine, only zero bytes are not read. TX of zero's is OK.
No handshaking is used.
Reading with: "res = hid_read(handle, buf, sizeof(buf));"
An actual string like A21B0E0300110A4B45 is read as A21B0E03.
So reading stops at the 00's.
I have no idea where to find this problem.
Any help or hint will be appreciated.
Thanks!

todbot, 2019-05-29:
what is the value of `sizeof(buf)`? This looks like a buffer sizing issue.

dekosan, 2019-05-29:
Size is 60. A21B0E03110A4B45 (without the zero's) is reading OK (A21B0E03110A4B45).

dekosan, 2019-06-12:
Solved, and the HIDAPI is OK. The problem in my prog occured where the total size of the readout string was taken in a wrong way. Now, each time I read a byte, I count up a counter which gets me the correct size. This is now used for parsing the data.

closed 2019-06-12 by dekosan