signal11/hidapi #434: windows os  with hid_read() return code -1

CaffeMrDe, 2019-06-25:
I solved connect problem and device is connected with hidapi sucessedfully. But, it occured the problem the Unable to read(). hid_read() api interface return the code -1.

The logs following:
Manufacturer String: BESCO
Product String: KSL81P304
Serial Number String: (1033)
Indexed String 1: BESCO
Unable to send a feature report.
Unable to get a feature report.
Unable to read()

Why? Please let us discuss together .

jonathancross, 2019-07-11:
This repo is no longer maintained, see #431 

In order to help keep that issue on the top, please do us a favor by closing your issue and reaching out on one of the other channels ([libusb mailing list](https://sourceforge.net/projects/libusb/lists/libusb-devel) or opening an [issue at libusb/hidapi](https://github.com/libusb/hidapi/issues)).

Thanks!

CaffeMrDe, 2019-07-17:
ok. Thanks.

closed 2019-07-17 by CaffeMrDe