signal11/hidapi #439: How can I just download HIDAPI.DLL?

Zaaphod, 2019-08-20:
I don't have visual studio or even know how it would work. I'm trying to get python easyhid.py  [https://github.com/ahtn/python-easyhid](https://github.com/ahtn/python-easyhid)  to work under windows and it seems to require HIDAPI.DLL .. but the dll is not included.  I have searched everywhere for this and I can't find a place to just download it.   Do I really have to get visual studio, install it, and learn how it even works, to build this DLL myself?  can someone PLEASE help me and just tell me where I can download it?  It would be extremely helpful.  I have been looking for a simple way to obtain this DLL for days now.   Trying to get this DLL to use easyhid is making it anything but easy.

Youw, 2019-08-20:
See https://github.com/libusb/hidapi/releases

Also, note, this repository is no longer maintained

Zaaphod, 2019-08-20:
Thank you VERY much!!  I would have never found it there!  I was on the libusb page but they only tell you how to download it from git clone git://github.com/libusb/hidapi.git which only is good for Linux.. it would be nice if they had a link to the windows zip file right under the Linux download instructions!   Thanks again!!!

Youw, 2019-08-20:
Since your issue is solved, please do close this ticket, to keep #431 at the top.

closed 2019-08-20 by Zaaphod