signal11/hidapi #440: Cannot enumerate bluetooth devices on Windows

linjingxiong, 2019-08-30:
Hi:
   Thanks for such great project. Recently, I am using hidapi to test my bluetooth. On Ubuntu, I can communicate with my bluetooth devices when I link myproject with libhidapi_hidraw.so, things go well. But when i turn to windows, something goes wrong, I have compiled hiapi.lib && hidapi.dll successfully,  and i run the application hid-test, but none of my bluetooth found. Is hidapi support bluetooth on windows? Or what can cause this issue ?

Qbicz, 2019-10-18:
Hi,

This is a question of support for Bluetooth devices on Windows.
1. Is it Bluetooth Low Energy or Bluetooth Classic? Bluetooth Low Energy is only supported on Windows 8 and Windows 10.
2. Is your BT device paired with computer? You can access HID only on paired Bluetooth devices. You need to first scan from computer, connect and pair, than you can run hidapi test program.

Qbicz, 2019-10-18:
Also, this project is abandoned. Please check https://github.com/libusb/hidapi which is actively maintained.