signal11/hidapi #442: hidDevice.open() reurn NULL

Chakib-Temal, 2019-10-28:
hello 

i use your libraries Hid4Java on My Ubuntu 16.04

all is fine to detect devices and read them parameters

but when i call 
        HidDevice hidDevice = hidServices.getHidDevice(6753, 13904, null);
it's a FreeStyle Libre device, and it's detectable with hidServices.getAttachedHidDevices()

i'm trying to find the problem , i'm sure it"s the Path
HidDevice [path=0001:0008:00, vendorId=0x1a61, productId=0x3650, serialNumber=null, releaseNumber=0x100, manufacturer=null, product=null, usagePage=0x0, usage=0x0, interfaceNumber=0]

but the function return Null , there is any problem with the path path=0001:0008:00 ?

if i can't to connect to the device , i can't communicate with him 

any suggestion ? Thank's a lot

Youw, 2019-10-28:
what is the version of hidapi?

```
HidDevice [path=0001:0008:00, vendorId=0x1a61, productId=0x3650, serialNumber=null, releaseNumber=0x100, manufacturer=null, product=null, usagePage=0x0, usage=0x0, interfaceNumber=0]
```
How did you get this output?

> manufacturer=null, product=null

this output is extremely rare and usually means broken HID driver or broken device

Chakib-Temal, 2019-10-29:
for (int devices = 0 ; devices <  hidServices.getAttachedHidDevices().size() ;  devices++) {
            System.out.println(hidServices.getAttachedHidDevices().get(devices));
}

here is the output : 
HidDevice [path=0002:0004:00, vendorId=0x413c, productId=0x2003, serialNumber=null, releaseNumber=0x301, manufacturer=null, product=null, usagePage=0x0, usage=0x0, interfaceNumber=0]
HidDevice [path=0002:0003:00, vendorId=0x461, productId=0x4d81, serialNumber=null, releaseNumber=0x200, manufacturer=null, product=null, usagePage=0x0, usage=0x0, interfaceNumber=0]
HidDevice [path=0001:0008:00, vendorId=0x1a61, productId=0x3650, serialNumber=null, releaseNumber=0x100, manufacturer=null, product=null, usagePage=0x0, usage=0x0, interfaceNumber=0]

1-Mouse , 2- keyboard 3-FreeStyle libre(glucometer) 
i'm not sure that is a broken device , you cann see the output for the mouse and the keyboard 

i'm trying now to use the hidApiTester in the command line, but it's not very clear to start
https://github.com/todbot/hidapitester/issues/1
may be you can help me if you already used it before ?