signal11/hidapi #445: Not able to open device in dynamic mode

SteMMo33, 2019-11-12:
Hi all,
I wrote a simple app to send data to LED strip device.
Using the library statically linked there are no problems.
Then I wrote another program with the library dynamically linked.
The device is not opened.
This is my code:

`

       lwsl_user("Devices:\n");
	struct hid_device_info* pInfo = pHidEnumerate( 0, 0);
	if (pInfo){
		struct hid_device_info* p = pInfo;
		while(p){
			lwsl_user("> %04X:%04X '%s' '%S' '%s'\n", p->vendor_id, p->product_id, p->path, p->manufacturer_string, p->product_string);
			p = p->next;
		}
	}
	else
		lwsl_user("- empty\n");
	// hid_free_enumeration(pInfo);
	// Open the device using the VID, PID,
	// and optionally the Serial number.
	wchar_t sn[80]={0};
	handle = pHidOpen(0x20A0, 0x41E5, sn);
	lwsl_user("fnOpen=%p sn='%S'\n", pHidOpen, sn);
	if (handle==0){
		lwsl_user(".. Device not found\n");
		return CMDRET_ERR;
	}	
	lwsl_user("> device handle: %X\n", handle);
	return CMDRET_OK;
`

This is the output:

`

Pointers ok. Open device ..
hid_init: 0
Lista devices:
 20A0:41E5 '0001:0027:00' '(null)' '(null)'
 14C8:0005 '0001:0003:00' '(null)' '(null)'
 0BDA:481A '0003:0002:ff' '(null)' '(null)'
fnOpen=0xb5dcda39 sn=''
.. Device not found
 closeLedStrip handle=0 hLibrary=80848690
`
The device is enumerated but not opened.
Any idea?

Thanks, regards

SteMMo33, 2019-11-13:
Sorry, not correct.
I linked statically and I have the same problem.
If I run other app on the same machine it works   o_O
Almost the same code ..
Both of them enumerate the HID devices and both try to open the same VID + PID (included in the enumerated devices).

`	handle = hid_open(0x20A0, 0x41E5, NULL);
`

The outpus are:

>hid_init: 0
>Lista HID devices:
> 062A:4101 'MOSART Semi.' '2.4G Keyboard Mouse'
> 062A:4101 'MOSART Semi.' '2.4G Keyboard Mouse'
> 20A0:41E5 'Agile Innovative Ltd' 'BlinkStick'
> 14C8:0005 'Zytronic Displays Limited' 'Zytronic Touchscreen Controller'
> 0BDA:481A 'Generic' 'USB Audio'
>open handle: 81330680
>Manufacturer String [res:0]: Agile Innovative Ltd

vs

>hid_init: 0
>Lista HID devices:
> 062A:4101
> 062A:4101
> 20A0:41E5
> 14C8:0005
> 0BDA:481A
>open handle: 0
>.. Device not found

Any idea?

Youw, 2019-11-13:
What platform is it? (Linux?)
Did you perform `hid_init` first?

In any case, I think you should already have a solution: you have a working application - simply check what is the difference.

SteMMo33, 2019-11-13:
Yes, the platform is ASUS Tinkerboard with Linaro (Debian for ARM).
Yes, i performed hid_init and the row
> hid_init: 0

is the return code.
The code (apart the debug strings) is the same  :(

Youw, 2019-11-13:
BTW: consider updating to a new version.
See #373.

SteMMo33, 2019-11-14:
Discovered the problem!
The applications was run by root; instead the application that did not work is launched by www-data user.
Is there a way to allow to www-data user to open the device?

Youw, 2019-11-14:
That is a general UNIX question, e.g.: https://unix.stackexchange.com/questions/172197/give-group-permissions-to-specific-device