signal11/hidapi #449: unable to use hid in kivy python mobile application

lakshyacis, 2019-12-09:
As I am creating an app in kivy python using buildozer a wrapper for python-for-android where a buildozer.spec file needs to be maintained and I simply wrote hidapi as requirement but it seems not working when says import hid

Youw, 2019-12-09:
Please see #373.

---
You didn't provide you build log/error message/build environment/etc.
Not enough info to know what exactly help do you need.

lakshyacis, 2019-12-09:
ImportError: Unable to load any of the following libraries:libhidapi-hidraw.so libhidapi-hidraw.so.0 libhidapi-libusb.so libhidapi-libusb.so.0 libhidapi-iohidmanager.so libhidapi-iohidmanager.so.0 libhidapi.dylib hidapi.dll libhidapi-0.dll

As I got to know these cant be installed using pip as they are not in pypi as I tried the installation of these in linux system where they got installed using

sudo apt-get install libudev-dev libusb-1.0-0-dev libfox-1.6-dev

sudo apt-get install libhidapi-hidraw0

so How can I manage it in spec file do guide me

Youw, 2019-12-09:
> kivy python mobile application

are you trying to build/run you application on some mobile platform?

> sudo apt-get install libhidapi-hidraw0

this should be more than enough to run hidapi/python application on your host Linux system

lakshyacis, 2019-12-09:
> > kivy python mobile application
> 
> are you trying to build/run you application on some mobile platform?
> 
> > sudo apt-get install libhidapi-hidraw0
> 
> this should be more than enough to run hidapi/python application on your host Linux system

yes its working on linux fine but the thing is for mobile phone application I am using kivy and buildozer and buildozer uses its spec file for making apk and build libraries and as these requirements are not available on pypi so it cant be installed using pip so how can I manage I am little bit confuse

Youw, 2019-12-09:
You have to use Android NDK to build hidapi to get libhidapi-hidraw.so/libhidapi-libusb.so and package it into your APK.
To build libhidapi-libusb you'd also need to build libusb with same Android NDK first.

---
This one is entirely on you.

lukaslehnert, 2020-07-18:
> ImportError: Unable to load any of the following libraries:libhidapi-hidraw.so libhidapi-hidraw.so.0 libhidapi-libusb.so libhidapi-libusb.so.0 libhidapi-iohidmanager.so libhidapi-iohidmanager.so.0 [...]

Hey! i got exactly the same error on windows. What solved your problem? Its not clear to me at all.. spend a lot time with the hidapi lib now but no success. Would be amazing to get some help.

lukaslehnert, 2020-07-18:
> Hey! i got exactly the same error on windows. What solved your problem? Its not clear to me at all.. spend a lot time with the hidapi lib now but no success. Would be amazing to get some help.

>> and it seems like "hidraw" ist the problem but iam not sure what to do.

Collecting hidraw
  Using cached hidraw-0.1.tar.gz (1.4 kB)
Building wheels for collected packages: hidraw
  Building wheel for hidraw (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: 'c:\users\lukas lehnert\pycharmprojects\python-for-mbot-master\venv\scripts\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\L
ukas Lehnert\\AppData\\Local\\Temp\\pip-install-_mbe992c\\hidraw\\setup.py'"'"'; __file__='"'"'C:\\Users\\Lukas Lehnert\\AppData\\Local\\Temp\\pip-install-_mbe992c\\hidraw\
\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'
))' bdist_wheel -d 'C:\Users\Lukas Lehnert\AppData\Local\Temp\pip-wheel-v_glf5h6'
       cwd: C:\Users\Lukas Lehnert\AppData\Local\Temp\pip-install-_mbe992c\hidraw\
  Complete output (17 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win32-3.8
  creating build\lib.win32-3.8\hidraw
  copying hidraw\hidraw.py -> build\lib.win32-3.8\hidraw
  copying hidraw\__init__.py -> build\lib.win32-3.8\hidraw
  running build_ext
  building 'hidraw._hidraw' extension
  creating build\temp.win32-3.8
  creating build\temp.win32-3.8\Release
  creating build\temp.win32-3.8\Release\hidraw
  C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.26.28801\bin\HostX86\x86\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD "-Ic:\users\lukas lehn
ert\pycharmprojects\python-for-mbot-master\venv\include" "-IC:\Users\Lukas Lehnert\AppData\Local\Programs\Python\Python38-32\include" "-IC:\Users\Lukas Lehnert\AppData\Loca
l\Programs\Python\Python38-32\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.26.28801\include" "-IC:\Program Files (x86)\Window
s Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared"
"-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\winrt" "-IC:\Program Files (x86)\Windows K
its\10\include\10.0.18362.0\cppwinrt" /Tchidraw/hidrawmodule.c /Fobuild\temp.win32-3.8\Release\hidraw/hidrawmodule.obj
  hidrawmodule.c
  hidraw/hidrawmodule.c(2): fatal error C1083: Cannot open include file: 'sys/ioctl.h': No such file or directory
  error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\BuildTools\\VC\\Tools\\MSVC\\14.26.28801\\bin\\HostX86\\x86\\cl.exe' failed with exit status 2
  ----------------------------------------
  ERROR: Failed building wheel for hidraw
  Running setup.py clean for hidraw
Failed to build hidraw
Installing collected packages: hidraw
    Running setup.py install for hidraw ... error
    ERROR: Command errored out with exit status 1:
     command: 'c:\users\lukas lehnert\pycharmprojects\python-for-mbot-master\venv\scripts\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\
\Lukas Lehnert\\AppData\\Local\\Temp\\pip-install-_mbe992c\\hidraw\\setup.py'"'"'; __file__='"'"'C:\\Users\\Lukas Lehnert\\AppData\\Local\\Temp\\pip-install-_mbe992c\\hidra
w\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'
"'))' install --record 'C:\Users\Lukas Lehnert\AppData\Local\Temp\pip-record-vxurw9cn\install-record.txt' --single-version-externally-managed --compile --install-headers 'c
:\users\lukas lehnert\pycharmprojects\python-for-mbot-master\venv\include\site\python3.8\hidraw'
         cwd: C:\Users\Lukas Lehnert\AppData\Local\Temp\pip-install-_mbe992c\hidraw\
    Complete output (17 lines):
    running install
    running build
    running build_py
    creating build
    creating build\lib.win32-3.8
    creating build\lib.win32-3.8\hidraw
    copying hidraw\hidraw.py -> build\lib.win32-3.8\hidraw
    copying hidraw\__init__.py -> build\lib.win32-3.8\hidraw
    running build_ext
    building 'hidraw._hidraw' extension
    creating build\temp.win32-3.8
    creating build\temp.win32-3.8\Release
    creating build\temp.win32-3.8\Release\hidraw
    C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.26.28801\bin\HostX86\x86\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD "-Ic:\users\lukas le
hnert\pycharmprojects\python-for-mbot-master\venv\include" "-IC:\Users\Lukas Lehnert\AppData\Local\Programs\Python\Python38-32\include" "-IC:\Users\Lukas Lehnert\AppData\Lo
cal\Programs\Python\Python38-32\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.26.28801\include" "-IC:\Program Files (x86)\Wind
ows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared
" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\winrt" "-IC:\Program Files (x86)\Windows
[... cut ...]
Temp\\pip-install-_mbe992c\\hidraw\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(com
pile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\Lukas Lehnert\AppData\Local\Temp\pip-record-vxurw9cn\install-record.txt' --single-version-externally-manag
ed --compile --install-headers 'c:\users\lukas lehnert\pycharmprojects\python-for-mbot-master\venv\include\site\python3.8\hidraw' Check the logs for full command output.

lukaslehnert, 2020-07-18:
it seems like "hidraw" is the problem but i cannot find a workaround..

Collecting hidraw
  Using cached hidraw-0.1.tar.gz (1.4 kB)
Using legacy setup.py install for hidraw, since package 'wheel' is not installed.
Installing collected packages: hidraw
    Running setup.py install for hidraw: started
    Running setup.py install for hidraw: finished with status 'error'

    ERROR: Command errored out with exit status 1:
     command: 'C:\Users\Lukas Lehnert\AppData\Local\Programs\Python\Python38-32\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\LUKASL~1\\AppData\\Local\\Temp\\pycharm-packaging\\hidraw\\setup.py'"'"'; __file__='"'"'C:\\Users\\LUKASL~1\\AppData\\Local\\Temp\\pycharm-packaging\\hidraw\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\Lukas Lehnert\AppData\Local\Temp\pip-record-6tpwu_i_\install-record.txt' --single-version-externally-managed --compile --install-headers 'C:\Users\Lukas Lehnert\AppData\Local\Programs\Python\Python38-32\Include\hidraw'
         cwd: C:\Users\LUKASL~1\AppData\Local\Temp\pycharm-packaging\hidraw\
    Complete output (19 lines):
    running install
    running build
    running build_py
    creating build
    creating build\lib.win32-3.8
    creating build\lib.win32-3.8\hidraw
    copying hidraw\hidraw.py -> build\lib.win32-3.8\hidraw
    copying hidraw\__init__.py -> build\lib.win32-3.8\hidraw
    warning: build_py: byte-compiling is disabled, skipping.
   
    running build_ext
    building 'hidraw._hidraw' extension
    creating build\temp.win32-3.8
    creating build\temp.win32-3.8\Release
    creating build\temp.win32-3.8\Release\hidraw
    C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.26.28801\bin\HostX86\x86\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD "-IC:\Users\Lukas Lehnert\AppData\Local\Programs\Python\Python38-32\include" "-IC:\Users\Lukas Lehnert\AppData\Local\Programs\Python\Python38-32\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.26.28801\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\cppwinrt" /Tchidraw/hidrawmodule.c /Fobuild\temp.win32-3.8\Release\hidraw/hidrawmodule.obj
    hidrawmodule.c
    hidraw/hidrawmodule.c(2): fatal error C1083: Cannot open include file: 'sys/ioctl.h': No such file or directory
    error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\BuildTools\\VC\\Tools\\MSVC\\14.26.28801\\bin\\HostX86\\x86\\cl.exe' failed with exit status 2
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'C:\Users\Lukas Lehnert\AppData\Local\Programs\Python\Python38-32\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\LUKASL~1\\AppData\\Local\\Temp\\pycharm-packaging\\hidraw\\setup.py'"'"'; __file__='"'"'C:\\Users\\LUKASL~1\\AppData\\Local\\Temp\\pycharm-packaging\\hidraw\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\Lukas Lehnert\AppData\Local\Temp\pip-record-6tpwu_i_\install-record.txt' --single-version-externally-managed --compile --install-headers 'C:\Users\Lukas Lehnert\AppData\Local\Programs\Python\Python38-32\Include\hidraw' Check the logs for full command output.