signal11/hidapi #450: Unable to open device (access denied)

akiller, 2020-01-04:
Hi,

I'm attempting to access my keyboard to configure the RGB lights etc using ([DuckyAPI](https://github.com/Latedi/DuckyAPI)) that makes use of your library.

I have One2 TKL which is different model of keyboard to the one this project was targeted to but the same manufacturer. I've updated to the correct PID for my device and and am hoping they use the same protocol for all their devices. 

When I try and open my device hid_read_timeout() in hid.c executes `res = ReadFile(dev->device_handle, dev->read_buf, dev->input_report_length, &bytes_read, &dev->ol)` which is returning 0. Checking GetLastError() returns 5 (ERROR_ACCESS_DENIED). I'm running in an elevated administrator process.

Just to confirm I'm getting this same result when I compile the master branch of hidapi directly, running entirely separate from the DuckyAPI project.

This is the output from hidtest:

```Device Found
  type: 04d9 0356
  path: \\?\hid#vid_04d9&pid_0356&mi_02&col04#7&1ca74182&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
  serial_number: (null)
  Manufacturer: DuckyChannel International Co., Ltd.
  Product:      Ducky Keyboard
  Release:      120
  Interface:    2

Device Found
  type: 04d9 0356
  path: \\?\hid#vid_04d9&pid_0356&mi_02&col01#7&1ca74182&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: (null)
  Manufacturer: DuckyChannel International Co., Ltd.
  Product:      Ducky Keyboard
  Release:      120
  Interface:    2

Device Found
  type: 04d9 0356
  path: \\?\hid#vid_04d9&pid_0356&mi_02&col02#7&1ca74182&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: (null)
  Manufacturer: DuckyChannel International Co., Ltd.
  Product:      Ducky Keyboard
  Release:      120
  Interface:    2

Device Found
  type: 04d9 0356
  path: \\?\hid#vid_04d9&pid_0356&mi_00#7&28baf2ff&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
  serial_number: (null)
  Manufacturer: DuckyChannel International Co., Ltd.
  Product:      Ducky Keyboard
  Release:      120
  Interface:    0

Device Found
  type: 04d9 0356
  path: \\?\hid#vid_04d9&pid_0356&mi_02&col03#7&1ca74182&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: (null)
  Manufacturer: DuckyChannel International Co., Ltd.
  Product:      Ducky Keyboard
  Release:      120
  Interface:    2

Device Found
  type: 04d9 0356
  path: \\?\hid#vid_04d9&pid_0356&mi_01#7&4e3b53d&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: (null)
  Manufacturer: DuckyChannel International Co., Ltd.
  Product:      CY040356014C0102 EU
  Release:      120
  Interface:    1

Unable to open device```

Hopefully you can point me in the right direction.

Thanks.

Youw, 2020-01-05:
Please see #373

akiller, 2020-01-05:
Sorry I must have missed this wasn't being maintained. Thanks I'll try that instead.

closed 2020-01-05 by akiller