signal11/hidapi #454: failed to open the device

alireza11048, 2020-03-07:
Hi,

I am trying to communicate with an HID device with this library, I have two problems:
1. to open the device for write&read I need to run the app with admin permission, is this normal?
2. some times the library returns an error when I try to open the device, particularly it fails during calling CreateFileA function. and getlasterror() returns permision_denied in this situation. there is any known issue with this? and also the device path of my device is "\\\\?\\hid#vid_1915&pid_520c&mi_01#8&20c16deb&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd" and also it appears as keyboard in system too.

Youw, 2020-03-08:
Please see #373

Youw, 2020-03-08:
Additionally: Windows driver subsystem owns all mouses and keyboards, so you cannot open those for read/write from user-space.
That is a known limitation.
This is one of the items we should add as part of #61

alireza11048, 2020-03-09:
Thank you, the problem has solved.

closed 2020-03-09 by alireza11048