signal11/hidapi #458: Hardened runtime using hidapi under macOS Catalina

jbolduc01, 2020-05-13:
Hello, 
I am having trouble to codesign my QT app using macdeployqt using the hidapi project.
Seems like I do not have access to HID device after codesigning my application. 
Is anyone ran into this issue? How to solve it?
Thanks,
Jérôme

todbot, 2020-05-13:
This library isn't being maintained any more, please see #373. 
However the solution to your issue is your app needs extra entitlements.
See examples like: https://github.com/UltimateHackingKeyboard/agent/blob/master/scripts/entitlements.mac.plist
But I believe the most important entitlement is 
  `com.apple.security.cs.allow-dyld-environment-variables`
See:
https://developer.apple.com/documentation/bundleresources/entitlements/com_apple_security_cs_allow-dyld-environment-variables

Youw, 2020-05-13:
> com.apple.security.cs.allow-dyld-environment-variables

I believe it is not required any more after https://github.com/libusb/hidapi/pull/120

todbot, 2020-05-13:
Interesting! I will update `node-hid` and schedule some `node-hid` & Electron tests.  Hopefully this eliminates the need for this entitlement for Electron users too. Thanks!

jbolduc01, 2020-05-13:
Fixes [#119](https://github.com/libusb/hidapi/issues/119) solves my issue on my side. Thanks.