signal11/hidapi #480: Request for help implementing HID protocol in BrailleTouch project

discapacidad5, 2023-04-09:
hazme uno dirigido  persona en que ce trabajan con el protocolo HID

Youw, 2023-04-09:
Please see https://github.com/signal11/hidapi/issues/468 and close this issue.

discapacidad5, 2023-04-09:
> Consulte [el n.º 468](https://github.com/signal11/hidapi/issues/468) y cierre este problema.

Dear Youw.

I came across your HIDAPI library for Windows, Linux, FreeBSD, and Mac OS X while researching how to interface with USB and Bluetooth HID-Class devices. I noticed that you have experience working with the HID protocol, and I was wondering if you could offer some guidance.

Specifically, I am trying to implement HID Braille functionality in my project, which involves sending and receiving Braille input, output, and feature reports. I understand that you have not worked with Braille devices specifically, but I am hoping that your experience with HID will be of use.

I apologize if this request is outside the scope of your work or if you are not available for consultation. I understand that your HIDAPI repository is no longer maintained, but I am hoping that you may still be able to offer some guidance.

Thank you for your time and consideration.

Best regards,

Cergio Monasterio